d-ronin/dRonin #1985: RFC: Python 3

tracernz, 2017-10-27:
Can we drop Python 2? It's history and annoying to support. If not for the libs yet (who doesn't have 3 though, seriously??), at least for the scripts in the build/packaging system?

mlyle, 2017-10-27:
Is there any standard way to guarantee a python3 invocation? :/

tracernz, 2017-10-28:
Call python3 instead of python/python2 like we currently do. That works on win/mac/linux for me.

For scripts outside the build system, `#!/usr/bin/env python3` should work, including windows (python launcher on Windows was fixed to handle shebang properly years ago afaik?).

mlyle, 2017-10-28:
I believe arch helpfully does not have /usr/bin/python3, and /usr/bin/python is python3.  :P   That's why I've made both work so far and have it pointing to python2 on most things.

Would love to just support 3; it's been a huge cause of regressions.  (Though I have some things to relearn).

tracernz, 2017-10-28:
> I believe arch helpfully does not have /usr/bin/python3, and /usr/bin/python is python3. :P 

Maybe historically? Fortuantely historical isn't something you ever need to worry about with Arch. :grin: 
```
mike-desktop:~/dev/dronin $ uname -a
Linux mike-desktop 4.13.8-1-ARCH #1 SMP PREEMPT Wed Oct 18 11:49:44 CEST 2017 x86_64 GNU/Linux
mike-desktop:~/dev/dronin $ python3
Python 3.6.2 (default, Jul 20 2017, 03:52:27) 
[GCC 7.1.1 20170630] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

mlyle, 2017-10-28:
Nice, let's do it then.  Build tools first.  Will have to check / possibly install on windows builder.

tracernz, 2017-10-28:
Yeah, Windows is as usual the one I'm worried about.

tracernz, 2017-12-09:
Ubuntu are now building a 17.10 image _without_ python 2: https://lists.ubuntu.com/archives/ubuntu-devel-announce/2017-December/001234.html.

PR #2109 (2018-03-11): scripts, api: move to python3
Fixes #1985 (or will, when this is finished).

Need to test
* [x] Build machine: Linux
* [x] Build machine: Mac
* [x] Build machine: Windows
* [x] a/flight/Modules/CharacterOSD/encode_font.py
* [x] a/flight/Modules/OnScreenDisplay/inc/encode_fonts.py
* [x] a/flight/Modules/OnScreenDisplay/inc/encode_images.py
* [x] a/flight/targets/playuavosd/px_mkfw.py (tested by build)
* [x] a/flight/targets/playuavosd/px_uploader.py
* [x] a/make/scripts/version-info.py (tested by build)
* [x] a/python/dronin-canonicalize
* [x] a/python/dronin-dumplog
* [x] a/python/dronin-getco [... cut]
closed 2018-03-11 by mlyle