d-ronin/dRonin #1998: Trying to program BLHeli32 ESCs with passthrough

keithws, 2017-11-28:
* Seen in version: 1.3.1
* On flight controller: BrainFPV RE1
* With Spedix GS30 BLHeli_32 ESCs

Issue details:

When trying to [program ESCs with passthrough](http://dronin.org/docs/advanced-configurations/programming-escs-with-passthrough/), I get the following warning from BLHeliSuite32.

> Connection to 4-way Interface "m4wFCINtf" succeeded.
> 
> Found outdated interface revision 20000!
> 
> Please update the Interface to a supported revision!
> (Protocol 20001 or higher)

If I understand this correctly (which is a big if), then the interface provided by dRonin in the FC software needs to be updated. Yes?

tracernz, 2017-11-28:
Unfortunately this seems to be the way in that ecosystem. :roll_eyes: Change the version on our end and you break existing setups.

mlyle, 2017-11-28:
It's already changed, since a week after release.  They bumped it with no other code change.  Was a super-confused move on their part :P 

Nightlies etc work with current blheli_32, with all the caveats of prerelease software.

keithws, 2017-11-28:
Thanks for the quick response. So ignore the warning or use a nightly. Hmm... ⚖️

tracernz, 2018-01-24:
So to confirm, we're happy with this for Wired release?

mlyle, 2018-01-25:
Yes

closed 2018-01-25 by mlyle