d-ronin/dRonin #2008: Can't see apply/save buttons on hardware screen

ufoDziner, 2017-12-08:
* Seen in version (release version or githash): 6b208f+
* On flight controller (if applicable): Seppuku

Issue details:
 Apply/Save buttons at the bottom of the Hardware screen are not visible.

[image]

tracernz, 2017-12-08:
Does it scroll?

ufoDziner, 2017-12-08:
Not that I can see.

tracernz, 2017-12-08:
Did you try mouse wheel? Not sure if the buttons are missing or just the scrollbar has visibility issues.

ufoDziner, 2017-12-08:
Just verified.  No scrolling available via mousewheel.

tracernz, 2017-12-08:
Kinda repros in Linux/X11. The scrollbar is there, and mousewheel works, but the buttons are only half-visible when scrolled all the way down. Looks like the widget encompassing the config form extends to the bottom of the window underneath the tab bar.

PR #2009 (2017-12-08): GCS:Seppuku: Remove minimum size of config form
Let the scroll area do it's job. Fixes #2008
closed 2017-12-08 by mlyle