d-ronin/dRonin #2017: RGB LED count

ufoDziner, 2017-12-23:
* Seen in version (release version or githash): 
* On flight controller (if applicable): Seppuku

Issue details:
When there are no additional LEDs, the  Total LEDs listed are 0 on the RGB LEDs tab of the Modules section.  However, when additional LEDs are added (say 10 external), the Total LEDs listed must be set to 11 (I'm told to account for the one on the board).  If this box does include the onboard LED, it should default to 1, not 0.

Please set the default Total LEDs to 1, if it is to include the onboard LED, or have the Total LEDs only include the external count.  Thanks!

mlyle, 2017-12-23:
Unfortunately this isn't so easily done-- we don't have a mechanism to have different defaults for different boards.  And setting the value to a number higher than 0 has the effect of enabling RGB LEDs.  We just have some special code on seppuku to force the LED value to 1 if it's zero to ensure the onboard LED always works.

ufoDziner, 2017-12-23:
Can a note be added below the Total LEDs line that states:
(For Seppuku FC add 1 to the number of _external_ LEDs installed.)
?

yamahabest, 2017-12-26:
Isn’t it possible to just always do plus one in the software, for the Seppuku boards?

tracernz, 2017-12-26:
All of this code is totally generic. You could drop a Seppuku specific adjustment in there somewhere but it'd be a bit ugly. Maybe can do something better on the flight side...

PR #2061 (2018-02-12): GCS:Modules: Show a hint about RGB LEDs on the board
Fixes #2017
closed 2018-02-12 by mlyle