d-ronin/dRonin #2022: not displaying floats on OSD

droney, 2017-12-31:
* Seen in version (release version or githash): 79dbcf6f
* On flight controller (if applicable): seppuku

Issue details: all floats are not being displayed on the OSD. Instead, '<digit>f' is being displayed. This was observed on build 1558 in both the linux and windows builds. I did not try earlier builds.

droney, 2017-12-31:
I removed 'SRC += printf-stdarg.c' from flight/PiOS/pios_flight_library.mk and was able to successfully compile code for seppuku and floats are now being displayed correctly on the OSD.

tracernz, 2018-01-01:
Thanks for tracking it down! I wonder if we can use newlib-nano printf. I'm not sure we have implemented the re-entrant stubs so we may want to do that.

mlyle, 2018-01-01:
We used that before-- this is a regression from factoring makefiles common-- OSD targets don't want printf-stdarg.c because of its limitations.

droney, 2018-01-01:
I thought linker flag c_nano invoked newlib-nano?? I noticed c_nano is now being used during linking, which is different from the neat release.

tracernz, 2018-01-01:
> We used that before-- this is a regression from factoring makefiles common-- OSD targets don't want printf-stdarg.c because of its limitations.

Ah, that makes sense. 👍 

> I thought linker flag c_nano invoked newlib-nano?? I  noticed c_nano is now being used during linking, which is different from  the neat release.

I think the newlib symbols are overriden by anything else you bring in.

PR #2023 (2018-01-01): pios_flight_library: remove printf-stdarg.c
Instead include it on the individual non-OSD targets.

Thanks to @droney for the report.  Fixes #2022
closed 2018-01-01 by tracernz