d-ronin/dRonin #2069: Wired OSX Artifacts don't work

mlyle, 2018-02-13:
Unknown cause-- seems to be an intermittent issue with the OSX Builder.  Rebooted and re-launched build to see if clean bits come out.

mlyle, 2018-02-13:
Probably fixed by #2071

mlyle, 2018-02-13:
Fixed/worked around.  I don't like nondeterminism, but "both kinds" of artifacts work.

closed 2018-02-13 by mlyle

tracernz, 2018-02-13:
Will set up some automatic builds and to see if I can repro this in a different env.

tracernz, 2018-02-13:
Mac packing script that runs macdeployqt, touches rpath etc. runs twice in the first release attempt, haven't checked any others but that doesn't seem good.
```
$ curl -s http://jar.lyle.org:8080/job/dronin/1623/Nodes=osx/consoleText | grep -n './macdeployqt "${APP}"'
8841:./macdeployqt "${APP}" -verbose=2 -no-strip -always-overwrite -qmldir="$QMLDIR" -executable="${APP}/Contents/MacOS/crashreporterapp"
13187:./macdeployqt "${APP}" -verbose=2 -no-strip -always-overwrite -qmldir="$QMLDIR" -executable="${APP}/Contents/MacOS/crashreporterapp"
```

Also, this bug was closed in Qt 5.5, and it makes me nervous...
```
#pushd/popd: workaround QTBUG-46404 -- need to be next to macdeployqt's path

pushd "${QT_SDK_BIN_PATH}"
```