d-ronin/dRonin #2098: Display hover % throttle next to hangtime info

ufoDziner, 2018-03-07:
* Seen in version (release version or githash): 
* On flight controller (if applicable): 

Issue details:
Would be great to have the craft's hover % throttle displayed near the Max Power Add portion of the GCS, and include a tooltip that suggests the recommended Max Power Add.

tracernz, 2018-03-07:
What about just having autotune tune the max power add to suggested value, or at least offer to?

ufoDziner, 2018-03-07:
That would be cool to have it offer, and have the user be able to accept/reject.  Would still be nice to have it displayed as well for further incremental changes.

jhitesma, 2018-03-08:
I was going to suggest what @tracernz said - have it set this automatically.  This would probably cut down on a good bit of confusion and make sure users get a reasonably sane value to start with.

But...I also agree with @ufoDziner that we should still display it to help guide users when/if they change that setting manually.

PR #2103 (2018-03-16): hangtime: when enabling, calculate power add value
Use the hover throttle from autotune * .85, when the user checks the hangtime box.  Sanity checks the range.

This isn't exactly what #2098  stipulates, but it's a good way to do it IMO.

Fixes #2098
closed 2018-03-16 by tracernz