d-ronin/dRonin #2101: Need a warning or something for "Collective" on multirotor

mlyle, 2018-03-08:
At least two people have tripped over this so far.

tracernz, 2018-03-08:
Collective selected somewhere, or?

mlyle, 2018-03-09:
As an input channel-- it prevents throttle control.

tracernz, 2018-03-09:
That's rather a peculiar thing to select unless you actually have collective pitch (e.g. Curtis Youngblood quads).

mlyle, 2018-03-09:
Lots of people set e.g. all the channels to sbus and then map them, and we behave in a pretty unintuitive way then.

yds, 2018-03-10:
how about disabling **Collective** in **RC Inputs** if **Vehicle Type:** _Multirotor_ in **Mixer Settings**

mlyle, 2018-03-10:
No, because there are multirotors with collective.

tracernz, 2018-04-16:
Is this still valid with #2117? Seems you can't really accidentally configure it now.

mlyle, 2018-04-16:
#2117 helps mitigate it.  some people put the whole left column "sbus" and left it none, some people configured channels for everything because "why not".  But I think we should still consider doing more.

glowtape, 2018-04-17:
Move the collective channel to the bottom of the list, followed by a warning about helis and such?

PR #2208 (2018-05-20): GCS:Config: Add warning for collective channel assigned on non-heli
This warning appears when collective is assigned a type and channel number for anything that isn't a CP heli. Is that too broad?

Fixes #2101
Supersedes #2207

[image]
closed 2018-05-20 by mlyle