d-ronin/dRonin #2104: OmnibusF4SD(Revolution) bootloops

yds, 2018-03-10:
* Seen in version (release version or githash): Release-20180212.2
* On flight controller (if applicable): OmnibusF4SD a.k.a. V2 and above

Issue details: v89 of the bootloader works as far as booting into the Rescue mode with the GCS but goes into a bootloop with the _Wired_ Revolution target.

Tested with three different variants of OmnibusF4SD, one with the built in current sensor and two without.

mlyle, 2018-03-10:
Isn't it documented that we don't support this?

mlyle, 2018-03-10:
(e.g. it's a different target in BF than revolution/omnibusf4)

yds, 2018-03-11:
Wishful thinking on my part. I thought I read somewhere this was being worked on for the _Wired_ release.

Never mind. I might give this a shot once I get a dev environment set up.

mlyle, 2018-03-11:
Sorry-- if a PR shows up with support, or I get bored and do it one day, it may happen.  But for now...

closed 2018-03-11 by mlyle

mlyle, 2018-03-15:
Bought this: http://www.readytoflyquads.com/flip-32-f4-omnibus-v2-pro  it works.  What gives? :P

yds, 2018-03-15:
I bought that same one when it first came out and I have a clone of it that looks identical. As I type this I had them both bootloop when flashed with _Wired.2_ :(  I also have the [OmnibusF4SD](http://www.readytoflyquads.com/flip-32-f4-omnibus-v2-1020) without Current sensor -- same thing - v89 fixed the bootloader, but instantly bootloops right after the bootloader.

I have an ST-Link.. could you point me to some docs on how to hook things up so I can help debug this?

I'll be happy to PayPal you the money for that FC.  :)  I really appreciate all the time you put into this project, spending your own coin on RTFQ is too much!

mlyle, 2018-03-15:
> I'll be happy to PayPal you the money for that FC. :) I really appreciate all the time you put into this project, spending your own coin on RTFQ is too much!

haha that's not necessary.

It really seems to work for me, though I've not gone back to wired (but we've not changed anything that would "help" to my knowledge).  Of course, I expect output pin map will be wrong.

Using ST-Link on a random clone board isn't likely to work because they probably don't expose the pins / they may conflict with other uses.  The best thing to do is to look at pios_board.c and add panic lines with a strange number of flashes, like 4... and see how far you can move it down before you get to the "other issue".