d-ronin/dRonin #2110: windows build changes to python: document, announce

mlyle, 2018-03-11:
* [x] Docs:Install anaconda3
* [ ] Docs:Add a python.exe -> python3.exe symlink on windows (so f'n unfortunate)
* [ ] Docs: announce that a change is necessary

mlyle, 2018-04-20:
@glowtape or another windows expert, pretty pretty please?

tracernz, 2018-04-20:
Let's go to msys2 proper instead (rather than the gimped copy shipped with git for windows), and we can install git, python etc. with just `pacman -S python3 git`, and `/usr/bin/env python3` JustWorks™. Should be able to get openssl dlls we need from it to (installer is currently broken and I don't know how to fix it).

mlyle, 2018-04-20:
That sounds nice but whatever we instruct people on should match the CI server-- and I am hesitant to fix the CI server now because it feels like we're getting into shipping cadence (main remaining things that are a must are the windows regression and whatever I can fit in for openlrs-ness).

mlyle, 2018-05-16:
w/e

closed 2018-05-16 by mlyle