d-ronin/dRonin #2137: Hide OSD menu items

ufoDziner, 2018-03-21:
* Seen in version (release version or githash): 
* On flight controller (if applicable): 

Issue details:
Not everybody needs or uses all of the available items on the OSD menu.  It would be really nice to be able to show only specified items or groups of items.

mlyle, 2018-03-21:
Like, how?  Have configuration for which subsets?  Break things into more screens?

IMO it's better to have things there and available than be surprised you can't reach them.

ufoDziner, 2018-03-21:
What I'm currently imagining is on the OSD portion of the GCS is a tree style menu with checkboxes that determine which menus are visible on the OSD.  Just a 1 for 1 list of the line items/groups that are currently there.  So for like the VTx portion, it could be a group that enables power and channel, or they could just have channel available if they never change the power output.

All items could be checked by default to address your visibility concern.

ufoDziner, 2018-03-21:
So, if the person never uses TxPID, they could hide that entire portion.  I don't even know what other items are there, but I do know the list is long.  It makes sense to me to only show the relevant items for that pilot on the OSD menu rather than every item.

tracernz, 2018-03-21:
I've thought about making a UAVO browser in the OSD before. It'd take a good amount of flash to store all the strings. The problems are what limits to apply if there are none specified in the field def, what increments to step in, and others I haven't thought of yet. :stuck_out_tongue:

mlyle, 2018-03-21:
Yah, uavo browser would be cool.

I think this is WONTFIX, because it's difficult to do with the current OSD architecture, and a limited gain (makes the menu work slightly better if you invest some effort upfront).

closed 2018-03-21 by tracernz