d-ronin/dRonin #2143: SessionManaging rework proposal

mlyle, 2018-03-30:
1. GCS stores tristate for each object IsPresentOnHardware -- DONTKNOW, True, False.
2. At SessionManaging transition to connected, all are set DONTKNOW
3. FC SessionManaging enumeration only enumerates multi-instance objs, ignores other objs.
4. Receiving SessionManaging enumeration for obj sets isPresentOnHardware True.
5. Receiving any object sets isPresentOnHardware True
6. Receiving NAK for any object sets isPresentOnHardware False
7. 3times: GCS goes through a loop, through all the objs, and requests each that is currently DONTKNOW.  

For objects that are telemetered frequently, we get their state for free (0 roundtrips).  We only need one round-trip per settings object vs. 2 before (is present & req).

mlyle, 2018-03-31:
@tracernz, @glowtape, Any "oh shit don't do this" reasons?

tracernz, 2018-03-31:
No. Matches my thoughts, and seems like the right approach. :+1: Am away for the next few days, but I will have some time to look over anything. 😀

glowtape, 2018-03-31:
Sounds sane. I suppose reducing the round-tripping will reduce traffic on the connection and increase the chances of things making it through, improving things a lot for the flakier targets.

tracernz, 2018-03-31:
Exactly 👍

mlyle, 2018-04-18:
A more thorough version of this was done in #2148.

closed 2018-04-18 by mlyle