d-ronin/dRonin #2185: Buzzer inverted on HGLRC Zeus

lrvick, 2018-04-27:
* Seen in version (release version or githash): 0548c4791 
* On flight controller (if applicable): HGLRC F4 Zeus

Issue details:

Buzzer is inverted. If everything in annuciator config is set to "haironfire" then buzzer is on no matter what. When I set it to other modes like "OK" it does different beep codes depending on if radio is on or off, etc, however they are always directly opposite to the on/off state of LEDs with the same setting.

If someone can point me the right direction in the code I am happy to do some testing myself.

I tried setting ```active_high``` to false with no love: https://github.com/d-ronin/dRonin/blob/next/flight/targets/revolution/board-info/board_hw_defs.c#L117 

Should I actually be trying to invert the GPIO_Mode or GPIO_PuPd  defaults here?

mlyle, 2018-04-27:
How'd you wire the buzzer?

lrvick, 2018-04-28:
@mlyle I was positive I had my wiring correct per labeled pins, but not positive the vendor/firmware maps them correctly.

After some more trial and error changing annuciator options individually it turns out the pin labeled Buzzer/BB actually maps to "Alarm LED".

This would explain why flipping active_high had no impact.

I was able to get everything working properly by setting active_high=true on PIOS_LED_ALARM here:

```
diff --git a/flight/targets/revolution/board-info/board_hw_defs.c b/flight/targets/revolution/board-info/board_hw_defs.c
index a4c761e58..b6b01896b 100644
--- a/flight/targets/revolution/board-info/board_hw_defs.c
+++ b/flight/targets/revolution/board-info/board_hw_defs.c
@@ -82,6 +82,7 @@ static const struct pios_annunc pios_annuncs_v2[] = {
                                .GPIO_PuPd = GPIO_PuPd_UP
                        },
                },
+               .active_high = true,
        },
 };
```
Looks like a simple case of a pin mismatch on this board.

Still worth reopening to patch this in as a new board revision with a matching pios_annunc_v3_cfg ?

tracernz, 2018-04-28:
Yeah, I think the pinout came up in the IRC conversation.
@mlyle I think this is yet another board with a slightly different combo. Has mem chip, but with omnibus-ish pinout.

mlyle, 2018-04-28:
> Still worth reopening to patch this in as a new board revision with a matching pios_annunc_v3_cfg ?

Can we detect it in any way to do so?

lrvick, 2018-04-30:
Not sure. Any suggestions on probing the device for unique features or identifiers?