d-ronin/dRonin #2205: GCS: Build broken on GCC 8.1

tracernz, 2018-05-18:
Bad code in qsciencespinbox.. according to our source file:
```c++
/**
 * Code copied from http://www.matthiaspospiech.de/blog/2009/01/03/qt-spinbox-widget-with-scientific-notation/
 */
```

```
$ gcc --version
gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ make -j8 gcs
.... snip ...
compiling /home/mike/dev/dronin/ground/gcs/src/libs/qscispinbox/QScienceSpinBox.cpp
moc /home/mike/dev/dronin/ground/gcs/src/libs/qscispinbox/QScienceSpinBox.h
compiling .moc/debug-shared/moc_QScienceSpinBox.cpp
In file included from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qbrush.h:50,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qpalette.h:46,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qwidget.h:48,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qabstractspinbox.h:44,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qspinbox.h:44,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/QDoubleSpinBox:1,
                 from /home/mike/dev/dronin/ground/gcs/src/libs/qscispinbox/QScienceSpinBox.h:13,
                 from /home/mike/dev/dronin/ground/gcs/src/libs/qscispinbox/QScienceSpinBox.cpp:4:
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h: In member function ‘QTransform& QTransform::operator=(QTransform&&)’:
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h:81:46: error: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘class QTransform’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
     { memcpy(this, &other, sizeof(QTransform)); return *this; }
                                              ^
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h:57:20: note: ‘class QTransform’ declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
cc1plus: all warnings being treated as errors
make[4]: *** [Makefile:740: .obj/debug-shared/QScienceSpinBox.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qbrush.h:50,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qpalette.h:46,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qwidget.h:48,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qabstractspinbox.h:44,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/qspinbox.h:44,
                 from /home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtWidgets/QDoubleSpinBox:1,
                 from .moc/debug-shared/../../../../../../../../ground/gcs/src/libs/qscispinbox/QScienceSpinBox.h:13,
                 from .moc/debug-shared/moc_QScienceSpinBox.cpp:9:
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h: In member function ‘QTransform& QTransform::operator=(QTransform&&)’:
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h:81:46: error: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘class QTransform’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
     { memcpy(this, &other, sizeof(QTransform)); return *this; }
                                              ^
/home/mike/dev/dronin/tools/Qt5.9.2/5.9.2/gcc_64/include/QtGui/qtransform.h:57:20: note: ‘class QTransform’ declared here
 class Q_GUI_EXPORT QTransform
                    ^~~~~~~~~~
cc1plus: all warnings being treated as errors
make[4]: *** [Makefile:743: .obj/debug-shared/moc_QScienceSpinBox.o] Error 1
make[3]: *** [Makefile:54: sub-qscispinbox-make_first-ordered] Error 2
make[2]: *** [Makefile:48: sub-libs-make_first-ordered] Error 2
make[1]: *** [Makefile:47: sub-src-make_first-ordered] Error 2
make: *** [Makefile:291: gcs] Error 2
```

mlyle, 2018-05-18:
Bumped to Ludicrous-- can't do tools changes this late in inconceivable cycle (want to ship)

mlyle, 2018-05-18:
It looks like we have a few dubious memcpy's too but should wait for upstream tools to be ready for 8.1 to hunt 'em.

tracernz, 2018-05-18:
I can still build with clang (6.0.0) without any hax (luckily we don't do -Werror there).