d-ronin/dRonin #2212: Ubuntu install broken on 18.04 - requires unavailable dependency (libjasper1)

reubenfirmin, 2018-05-28:
* Seen in version (release version or githash): Release-20180212.2

Issue details:
The Ubuntu packaged deb requires libjasper1, which is not available in the Ubuntu 18.04 repos. Downloading it manually from 17.10 repos & installing works (for now). Unclear why libjasper1 isn't available in 18.04, but build deps should likely be changed.

tracernz, 2018-05-28:
Thanks for the report. That dep is from Qt's support for JPEG2000 (https://github.com/qt/qtimageformats/blob/5.9/src/plugins/imageformats/jp2/qjp2handler.pri). Not sure if we can disable that. Maybe we can just delete the jp2 imageformat plugin from our package.

tracernz, 2018-05-28:
The relevant Ubuntu Qt packages:
https://packages.ubuntu.com/xenial/qt5-image-formats-plugins
https://packages.ubuntu.com/bionic/qt5-image-formats-plugins
They removed the jasper dep https://salsa.debian.org/qt-kde-team/qt/qtimageformats/commit/217ae80e8b6a650f2aa9147b105b1d81e1e53d1e

mlyle, 2018-05-29:
Any easy way to fudge / remove the dependency at deb-build-time?

tracernz, 2018-05-29:
@reubenfirmin Can you please test the deb in http://jar.lyle.org:8080/job/dronin/1798/Nodes=master/ ? Thanks.

reubenfirmin, 2018-05-29:
That worked fine, and all appears functional. Thanks!

closed 2018-05-29 by mlyle

tracernz, 2018-05-29:
Thanks for checking back, appreciate it. :+1:

PR #2213 (2018-05-29): Package:GCS: Remove libqjp2.so - JPEG2000 support
It has a dependency on libjasper which is unmet on recent Debian based systems, due to security issues.

Fixes #2212