d-ronin/dRonin #2214: Enabling FrSKY sPort Non Inverted on seppuku causes bootloop

LitterBugs, 2018-06-02:
* Seen in version (release version or githash):  Pre-Inconceivable 4a4727f2
          http://jar.lyle.org:8080/job/dronin/1800/Nodes=master/
* On flight controller (if applicable): Seppuku

Issue details:
Bootloop occurs when enabling "FrSKY sPort Non Inverted" on uart 1 or 4.  When enabled, and R-XSR RX telemetry is plugged into uart1/4, Seppuku will bootloop.  Unplugging telemetry pin stops the bootloop, but does not allow telemetry.  Reverting back to wired stops the bootloop and telemetry works again.  Tested with two different R-XSR RXes and two different Seppukus.  (beta and released)

LitterBugs, 2018-06-02:
Upgraded Screech build, and it works fine.  Exported Screech settings and imported into the F450 build and that build now works.  Reset F450 back to defaults, and reconfigured, does not work again.  I've also tried all UARTs now, and it affects them all

Trying to isolate difference in UAVO between defaults and Screech settings that cause the bootloop when enabling sPort Non Inverted.

LitterBugs, 2018-06-02:
Simplified testing down to 
1) Flash Wired.2, enable s.bus and sPort un-inverted.  
2) Upgraded to pre-inconceivable.
3) still works
4) deleted settings partition
5) enabled s.bus and sPort un-inverted.
6) doesn't work
Comparing uavo exports from both....

LitterBugs, 2018-06-02:
Looks like GCS is setting wrong value.  Diff shows:
<             <field name="Uart4" values="S.Bus Non Inverted"/>
---
>             <field name="Uart4" values="FrSKY SPort Non Inverted"/>

yds, 2018-06-02:
Reproduced on the [FrSky XSRF4O](https://www.FrSky-RC.com/product/xsrf4o/) -- an OmnibusF4SD clone with integrated XSR internally wired like so:

- **MainPort**: _S.Bus_
- **RxPortUsart**: _FrSky SPort Non Inverted_

on _Wired.2_ both of the above settings work.

setting _FrSky SPort Non Inverted_ results in a bootloop on the _Pre-Inconceivable_ 4a4727f http://jar.lyle.org:8080/job/dronin/1800/Nodes=master/

LitterBugs, 2018-06-02:
So narrowed it down to having battery monitoring enabled or not.  If FrSKY telem is enabled without battery monitoring enabled, the bootloop occurs.

tracernz, 2018-06-09:
Fixed in #2215

closed 2018-06-09 by tracernz