d-ronin/dRonin #2216: AltitudeHold flight mode inoperative

LitterBugs, 2018-06-04:
* Seen in version (release version or githash):  Pre-Inconcievable and #2215 
* On flight controller (if applicable): Seppuku

Issue details:
Selecting Altitude hold mode causes motors to drop to Neutral, Throttle is ineffective, And beeper sounds .-. "R" code.

Also noticed that there is still an ALT.H indicator on the Flight Data screen in GCS although there is no longer an AltitudeHold Module.  

If FC is connected to GCS when selecting Altitude Hold mode, the RC Input indicator turns red.  Hovering over it gives the message: "Error Code: Undefined      This should not happen"

glowtape, 2018-06-05:
When I define a Stab1/2/3 with AltitudeHold as thrust mode, it seems to work fine. When I specify AltHold as flight mode, it goes into failsafe. Gonna look into it, must be a flight snafu with the flight mode config.

tracernz, 2018-06-05:
Thanks for tracking this down @glowtape :+1: .

mlyle, 2018-06-06:
Introduced by me in 3dc6eb0ad9 when I removed duplicated code (but a little too much) added in 7ca566633d.  Sorry :/

tracernz, 2018-06-06:
No prob :wink:

PR #2217 (2018-06-06): transmitter_control: Don't forget about Altitude Hold.
Fixes #2216.
closed 2018-06-06 by tracernz