d-ronin/dRonin #2239: Python API:

raemin, 2019-01-11:
* Seen in Main Branch Scripts as of 21/12/2018 
* Logs produced by firmware 2238
* On flight controller Seppuku

Issue details: Can log older logs (presumably wired), but Inconceivable logs crash the logviewer.

Telemetry.py , error line 1017, no header signature

[Logs.zip](https://github.com/d-ronin/dRonin/files/2748898/Logs.zip)

[image]

raemin, 2019-04-09:
I've reverted back to wired, made a clean install, generated the log and could read them with the "pip installed" logviewer. No crash, nice graphs.

From then on I upgraded the firmware to inconceivable (2126), made another log and attempted to read it with the python scripts hosted on github (branch next as of april 8th). The viewer does not crash anymore, but all the graphs are empty.

See corresponding sample logs and screenshots.

[SampleLogs.zip](https://github.com/d-ronin/dRonin/files/3060754/SampleLogs.zip)

raemin, 2019-04-10:
Commenting again this issue ...
The problem seems to be related to the latest python source! In fact I was attempting to read wired logs with wired scripts and inconceivable logs with inconceivable scripts. It seems wired python scripts are capable to read both wired/inconceivable logs, but inconceivable scripts do only produce blank graphs.

If you take the samples provided in my previous post, all the logs the log can be viewed using the python scripts from the "wired" branch.

[image]

there are a few warnings though

[image]

Now the same "inconceivable" log as displayed by the "next" python scripts (as of April 8th)

[image]

No warning in the console with these latest scripts.

raemin, 2019-04-10:
Last but not least on my original post, it was firmware 2038 night build (not 2238 sorry for the typo). Latest investigations made on a clean wired install and inconceivable 2126 night build. By clean install I mean: dump the settings, edit the xml file, to keep only the relevant PIDs / inputs / actuators, / etc... wipe the flight controller partition and reload. This was to get rid of my custom metadata.

tracernz, 2019-04-11:
Thanks for the report, and your work to narrow it down. When time permits I can have a look at this.