d-ronin/dRonin #2241: GCS: Welcome tab "GitHub Activity" and Flight Data "MAP" not displaying on Ubuntu 18.04

LitterBugs, 2019-03-12:
* Seen in version (release version or githash): ALL
* On flight controller (if applicable): ALL

Issue details:
When using the latest Ubuntu 18.04 LTR release, no data is displaying in the Welcome Tab "GitHub Ativity" area.  The Map tiles on the Flight Data tab are also not displaying when using the  standard Google Sat maps.  They display fine when switching to "default"
[image]
[image]

tracernz, 2019-03-12:
I see the no GH activity with Qt 5.9.2 builds, and it is fixed with Qt 5.12.1 builds. The Map works either way for me on Arch. Will need to investigate that one.

glowtape, 2019-03-13:
An unscientific look at the progress bar in the lower right corner, considering the time it needs to fill up, and what the system monitor says in regards to network traffic, the tiles appear to be downloaded. Something's upsetting the map control, maybe something about the image format?

LitterBugs, 2019-05-07:
Interesting discovery tonight while trying to get GCS running on an fresh Windows7 box tonight.  Almost exact same situation occurred when getting MSVCR120.dll errors.  No Github activity or Map tiles.  That was remedied by installing the MS C++ 2013 runtime "vcredist_x86.exe".  Wonder if there is a similar library missing in the latest ubuntu.