d-ronin/dRonin #2242: GCS Input - RSSI: Can we / how could I add "iBus" option?

fxbisto, 2019-03-17:
All current versions, all flight controllers
Issue details:
As the issue states, could we make this happen?
I am using the 'FlyPlus' firmware for my FS-i6 transmitter which allows sending RSSI over an aux channel of your choice. Normally I have this set to channel 12 / aux 8 but there's no ibus option for this (presumably because it didn't support it in 2016)

A little background on me:
 I'm a bit of a power user. Most of my desktops are on Linux. My coding isn't the greatest but I can make easily make minor changes to source and compile stuff on my own when I need to: Ex. - remapping the S1 output on my board to the pad marked RX4 as I ripped S1 off during a crash a while back (when I was on Butterflight it was easy to remap via CLI) 
Removed RX4 assignment and enabled the pad as motor 1 output by hunting down the relevant section for my board config file and recompiled the target, then flashed the updated firmware.

I have a lot of free time at the moment as I'm off work due to a recent surgery, and would be happy to get stuck in on my own if someone who knows how could point me in the right direction to update xml's and other source files.

tracernz, 2019-03-17:
I saw your messages on IRC, but you had quite already. I'm a little hazy on how that works, it's been a while since I worked on RSSI or wrote the IBus driver. 

Places that definitely need additions:
https://github.com/d-ronin/dRonin/blob/3a866925d6148e33e0620c94c5940413574a7247/flight/Modules/ManualControl/transmitter_control.c#L119
https://github.com/d-ronin/dRonin/blob/3a866925d6148e33e0620c94c5940413574a7247/shared/uavobjectdefinition/manualcontrolsettings.xml#L37

Of interest:
https://github.com/d-ronin/dRonin/blob/3a866925d6148e33e0620c94c5940413574a7247/flight/Modules/ManualControl/transmitter_control.c#L329

Does the RSSI come through as a channel, or a dedicated field in the frame? Should be easy for the former. For the latter you could either fake a channel in the driver, or implement a method that transmitter_control.c can call to get RSSI, and add the necessary bit in there.

fxbisto, 2019-03-17:
Thanks for your reply.

Great work on the driver by the way, it works perfectly :) I don't think we'll need to delve into that code specifically because the controller does the hard work for you.

Once the brilliant FlyPlus firmware has been flashed to the Tx it enables you (amongst other benefits - timers, alarms, etc) to simply select 'ERROR' as a source for the aux channels of your choice. Exactly how you would normally assign a switch or VRA/B to the AUX channels. It also enables transmission of AETR+12aux assignable channels without added latency (sourced from https://github.com/benb0jangles/FlySky-i6-Mod- )
As you said, RSSI comes through as a channel.
Setting an AUX to use ERROR modifies the output of the channel between 1000uS-2000uS based on the RSSI calculated by the FS-i6 internally. 0% RSSI = 1000uS and 100% = 2000uS.

All we would need to do is enable the option within dRGCS to allow iBus input with a selectable channel, identical to how a standard PWM input would be read but over iBus instead of using a dedicated DAC and breaking out a wire from the Rx.

I did try adding an entry in manualcontrolsettings.xml but it caused the GCS to fail during compile as there are of course other dependencies and iBus wasn't defined elsewhere as an RssiType

tracernz, 2019-03-17:
Ahh, it probably needs added to a switch statement or two in GCS. I'll take a look tonight.

tracernz, 2019-03-18:
Does this patch do the job?
```patch
diff --git a/ground/gcs/src/plugins/config/inputchannelform.cpp b/ground/gcs/src/plugins/config/inputchannelform.cpp
index 47c4785d2..512dc3b56 100644
--- a/ground/gcs/src/plugins/config/inputchannelform.cpp
+++ b/ground/gcs/src/plugins/config/inputchannelform.cpp
@@ -210,6 +210,9 @@ void inputChannelForm::groupUpdated()
         case ManualControlSettings::RSSITYPE_TBSCROSSFIRE:
             count = 12;
             break;
+	case ManualControlSettings::RSSITYPE_IBUS:
+	    count = 10;
+	    break;
         case ManualControlSettings::RSSITYPE_NONE:
             count = 0;
             break;
diff --git a/shared/uavobjectdefinition/manualcontrolsettings.xml b/shared/uavobjectdefinition/manualcontrolsettings.xml
index 75bc2754f..97a08b6b1 100644
--- a/shared/uavobjectdefinition/manualcontrolsettings.xml
+++ b/shared/uavobjectdefinition/manualcontrolsettings.xml
@@ -46,6 +46,7 @@
         <option>OpenLRS</option>
         <option>FrSkyPWM</option>
         <option>TBS Crossfire</option>
+	<option>IBus</option>
       </options>
     </field>
     <field defaultvalue="0" elements="1" name="RssiChannelNumber" type="uint8" units="channel">
```

fxbisto, 2019-03-18:
I've just got back home so I'll fire up the machine, patch and report back.
This should require recompiling of the target as well shouldn't it - or am
I incorrect in thinking that the UAVOs are being updated to enable this?

Cheers

tracernz, 2019-03-18:
That's correct. You'll need to add a case here first, at a minimum: https://github.com/d-ronin/dRonin/blob/3a866925d6148e33e0620c94c5940413574a7247/flight/Modules/ManualControl/transmitter_control.c#L119

fxbisto, 2019-03-18:
Success! I've added the relevant information where you suggested and a new case for ibus in dRonin/flight/Modules/ManualControl/transmitter_control.c and we are in business. Do you have the time to create a PR or should I have a go?

tracernz, 2019-03-19:
Great work. Have a go! 😁 GitHub docs: https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

fxbisto, 2019-03-20:
Thanks for all your help. I may be suggesting a few other minor changes for consideration in the near future as I've been playing with CharacterOSD.
So far I've managed to work out how to call the temperature data from barometer (someone wrote placeholder for Temp but it displayed nothing), shift some formatting about, cheer up the boot screen a little with some simple personalisation, and add a new average cell voltage option because I find it so much easier to compare voltage sag between 4/5/6s batteries that way :)
[image]
[image]

I may start on the fork tomorrow if I get a moment to stop!
Cheers again

glowtape, 2019-03-20:
More people flying dR and flaunting the LQG mode. Pressure to finally implement the ESC autotelemetry for torque is rising.

fxbisto, 2019-03-21:
Not so much 'flaunting' per sei, but I was intrigued to try out an alternative control system for sure.
Quad was a 'little' too noisy for good performance so I haven't had much to shout about ;)
/offtopic

PR created. Thanks @tracernz 
https://github.com/d-ronin/dRonin/pull/2246

closed 2019-03-21 by fxbisto