d-ronin/dRonin #2253: Output max resets center

ufoDziner, 2019-04-12:
* Seen in version (release version or githash):  Current
* On flight controller (if applicable): 

Issue details:
I noticed that when changing the min and max values on the output tab, changing the max settings resets the center position to the min number.  However, changing the min does not affect the center value.

It would be great to have the center number stay for a few seconds or so when changing the max value.  If the center is out of the range after the max is set and the time elapses, it could be reduced to the average of the min/max or something.  Thanks

fxbisto, 2019-04-12:
I've noticed this too, my fix is to either use cut/paste or arrow keys.
Your suggestion would certainly make things more user-friendly though, it took me a while to come to that idea. The whole thing could do with an overhaul really, all this update rate nonsense should be servo-specfic in my opinion, or at very least be named in line with protocols and some defaults applied.

fxbisto, 2019-04-12:
PR submitted to fix the slider resets :)

tracernz, 2019-04-12:
This is when you're typing in max, because while you are typing the value is smaller... we should connect the slot responsible for limiting the neutral to the [editingFinished signal](https://doc.qt.io/qt-5/qlineedit.html#editingFinished), whereas currently I suspect it's connected to [textChanged](https://doc.qt.io/qt-5/qlineedit.html#textChanged).

>I've noticed this too, my fix is to either use cut/paste or arrow keys.
Your suggestion would certainly make things more user-friendly though, it took me a while to come to that idea. The whole thing could do with an overhaul really, all this update rate nonsense should be servo-specfic in my opinion, or at very least be named in line with protocols and defaults applied, same as the choices available in the setup wizard for example. Continuity of the UI so it doesn't feel so confusing when you first try to setup a flight controller!

The update rate is not servo specific because that would limit the types of vehicles we can support. The only time you don't need it is with Oneshot or DShot ESCs, a lot of other ESCs need it. Remember, this is not just a mini quadcopter firmware. In terms of presets... yep, did that a couple of years back but after some yak shaving the branch rotted.

tracernz, 2019-04-12:
The offending line: https://github.com/d-ronin/dRonin/blob/next/ground/gcs/src/plugins/config/outputchannelform.cpp#L367

fxbisto, 2019-04-12:
> The update rate is not servo specific because that would limit the types of vehicles we can support. The only time you don't need it is with Oneshot or DShot ESCs, a lot of other ESCs need it. Remember, this is not just a mini quadcopter firmware.

yes of course, I appreciate that completely - what I meant was making it more user-friendly like how the DSHOT options are in there, adding 125,42,MS too and relegating the more custom refresh rates to only show on an output channel assigned to a servo. Standard and fast PWM would surely cover 99.9% of dinosaur ESCs people are likely to be using?

> The offending line: https://github.com/d-ronin/dRonin/blob/next/ground/gcs/src/plugins/config/outputchannelform.cpp#L367

Doesn't this method stop the arrow buttons & keyboard from effecting immediately?