d-ronin/dRonin #2260: Checking valid ranges of parameters

KimHyungSub, 2021-07-22:
* Seen in version (release version or githash): 
Current master (adbada45fb42dc8dc83cc7984ae4fbb2153b01fa)

Issue details:

Dear all,

While I tested some configuration parameters, I noticed that dRonin GCS does not enforces the valid ranges of parameters (e.g., VelocityKp, VelocityKi, AccKp, AccKi, etc.). Is there any place get the valid ranges of the parameters? Also, It would be awesome if the GCS enforces the parameters to have values within the valid ranges like what other flight control software already did :) Thanks!

tracernz, 2021-07-22:
It's been a few years so I'm very rusty, but... The UAVObjects are defined in XML. Each field can have a `limits` attribute that specifies the acceptable limits, and I _think_ the UAV Object Browser will either stop, or highlight the value if you try to go outside the limits. Unfortunately the limits for the AltititudeHoldSettings UAV object were never specified by anyone: https://github.com/d-ronin/dRonin/blob/next/shared/uavobjectdefinition/altitudeholdsettings.xml. You can see an example of limits in https://github.com/d-ronin/dRonin/blob/next/shared/uavobjectdefinition/stabilizationsettings.xml#L16. The format for limits is somewhat documented in the GCS source code: https://github.com/d-ronin/dRonin/blob/next/ground/gcs/src/plugins/uavobjects/uavobjectfield.cpp#L136-L141

As for the limit values... they are PID gains, so there is no obvious limit.

KimHyungSub, 2021-07-22:
@tracernz Thanks for the feedback.
I just checked the xml files. This is nice!
Yes, I agree that limiting PID gains is not obvious.
I mean that it will be great if assigning abnormal values (e.g., -999999999) can be prohibited by both users and the GCS. FYI, This was my bad. But, I unintentionally assigned such a strange value to a parameter and spent much time to figure out what was the problem.

tracernz, 2021-07-23:
You could make the limits "BI-bigger" than 0 e.g. `limits="%BI:0"`, for each of the gains in https://github.com/d-ronin/dRonin/blob/next/shared/uavobjectdefinition/altitudeholdsettings.xml